christophediericx/ArduinoDriver #1: Command line arguments expect C:\Users\chris\Desktop\sketch.hex to exist

yeiacatl, 2016-09-22:
The Debug Start options in the SketchUploader project configuration expect the following command line arguments:

--port=COM3 --file="C:\Users\chris\Desktop\sketch.hex"

What is the sketch.hex file supposed to contain? Can you share it?

christophediericx, 2016-09-22:
In the ArduinoLibCSharp solution, the _SketchUploader_ project's main purpose is to upload an (arbitrary) .HEX file to an Arduino board directly. For example, if you toggle verbose output in the Arduino IDE options, you should be able to spot where the (compiled) .HEX file output is written to on the filesystem (which happens just prior to upload). You could copy this file somewhere, and you can then use the SketchUploader project in order to upload this .HEX file directly to the Arduino board - without having to use the IDE. In many ways, this would be similar to using the tool 'avrdude' from the command line (which is what the IDE is doing behind the scenes). So in short, if you want to test uploading a compiled sketch file (.HEX) to a board, just provide a valid path to any HEX file in the Debug Start options, and you'll be able to debug the upload process directly from within Visual Studio. Please let me know if you require further clarification or explanation.

christophediericx, 2016-09-26:
The .user file (which contains this -local- debug path preference) has been removed from source control. That should resolve potential confusion.

closed 2016-09-26 by christophediericx