christophediericx/ArduinoDriver #6: 1. Add Work Project on VS2010; 2. Do It Library works on Linux?

releyshic, 2016-12-15:
1. Add Work Project on VS2010; 2. Do It Library works on Linux?

christophediericx, 2016-12-18:
Hi releyshic,

Thank you for your interest. Regarding your questions:

1. I don't have a Visual Studio 2010 instance around at the moment, but it should be very easy to just start a VS2010 project, drag and drop the corresponding source files in the correct location and link them up in the IDE?

2. I have looked into porting this library to .NET Core (which -among other things- would allow it to be run on Linux). At the moment `System.IO.SerialPort` is not supported within .NET Core however (rather crucial), which would make this exercise substantially harder. Perhaps when and if support for that gets added to .NET Core later, I might consider porting this library again.

christophediericx, 2016-12-18:
Related: https://github.com/dotnet/corefx/issues/984

releyshic, 2016-12-20:
Maybe this library ([http://www.teuniz.net/RS-232/]) to solve the problem?

christophediericx, 2017-11-17:
Related: https://github.com/christophediericx/ArduinoSketchUploader/issues/19

DanWBR, 2018-01-02:
Couldn't this library run on Mono on Linux/macOS? https://github.com/mono/mono/blob/master/mcs/class/System/System.IO.Ports/SerialPort.cs