christophediericx/ArduinoDriver #7: Using Arduino Uno, but not r3

Erpiz, 2016-12-21:
Will this cause an error?

when I execute the sample code against my arduino uno, which does not explicitly say it is r3 (so i am assuming it isn't) it fails on the handshaking. The arduino works with the normal arduino IDE. 

is this driver compatible with the older uno?

christophediericx, 2016-12-29:
Sorry for the delay in responding to this issue.

I have never tested the library with a non R3 Arduino, but I would have suspected it would "just work" (if the boot loader in use is identical). What would be helpful in further diagnosing this problem:

1. The exact way avrdude is called from within the Arduino IDE (verbose logging needs to be enabled in order to see the exact command line parameters used).

2. The exact error message that is displayed when the handshaking fails.

That being said, this is likely an issue with the "auto-bootstrap" feature only. Could you please try programming [the listener code](https://github.com/christophediericx/ArduinoDriver/blob/master/Source/ArduinoDriver/ArduinoListener/ArduinoListener.ino) to the board manually (using the Arduino IDE), and then try re-running the sample project code please?

ProgenitorX, 2017-04-11:
In relation to this, I was trying your library with a Nano (not an R3) and was running into issues when creating the ArduinoDriver instance (IOException - "Device Error"). I followed your suggestion of turning off the bootstrapper and loading the listener code manually, and I was able to successfully connect and interact with the Nano as expected. Below is my avrdude verbose output:

         C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -CC:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf -v -patmega328p -carduino -PCOM7 -b57600 -D -Uflash:w:C:\Users\LASERR~1\AppData\Local\Temp\arduino_build_265166/ArduinoListener.ino.hex:i 

         avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM7
         Using Programmer              : arduino
         Overriding Baud Rate          : 57600
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 2
         Firmware Version: 1.16
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

         avrdude: AVR device initialized and ready to accept instructions

         Reading | ################################################## | 100% 0.00s

         avrdude: Device signature = 0x1e950f (probably m328p)
         avrdude: reading input file "C:\Users\LASERR~1\AppData\Local\Temp\arduino_build_265166/ArduinoListener.ino.hex"
         avrdude: writing flash (5838 bytes):

         Writing | ################################################## | 100% 1.67s

         avrdude: 5838 bytes of flash written
         avrdude: verifying flash memory against C:\Users\LASERR~1\AppData\Local\Temp\arduino_build_265166/ArduinoListener.ino.hex:
         avrdude: load data flash data from input file C:\Users\LASERR~1\AppData\Local\Temp\arduino_build_265166/ArduinoListener.ino.hex:
         avrdude: input file C:\Users\LASERR~1\AppData\Local\Temp\arduino_build_265166/ArduinoListener.ino.hex contains 5838 bytes
         avrdude: reading on-chip flash data:

         Reading | ################################################## | 100% 1.25s

         avrdude: verifying ...
         avrdude: 5838 bytes of flash verified

         avrdude done.  Thank you.

Hope this was helpful!

christophediericx, 2017-06-14:
Closing this issue, no further comments from the topic starter received.

closed 2017-06-14 by christophediericx