christophediericx/ArduinoDriver #9: Compatibility with 101

erunks, 2017-01-09:
I'm assuming it's not already in since I didn't see it on the readme, but if you could get support for the Arduino 101, that'd be awesome.

christophediericx, 2017-01-10:
Hi erunks,

Thank you for your interest in this library.

I will take some time (early next week, I have a very busy week now) to evaluate what it would take to add support. It's likely that support can be added (probably even quite easily), but the total turnaround time will probably be 2 - 3 weeks from now.

Thanks,

Christophe

christophediericx, 2017-01-19:
Hi erunks,

Quite the interesting platform (the 101). Quite different from the other Arduinos in the family. I can't add support for it immediately, but I will keep the ticket open to remind me of it. Hopefully I'll find some time to implement support somewhere in the next couple of months.

Thanks,

Christophe

erunks, 2017-01-19:
Hi Christophe,

I greatly appreciate the time you plan to spend on supporting this model of Arduino. There is no rush on this from me either, so you don't have to worry about fitting it in if you can't.

Thanks,

erunks