christophediericx/ArduinoDriver #10: 2 bug-fixes for 2.1.0 release..

ghingres, 2017-01-21:
Hi Chris.

  I've been testing your interface for use as a bridge between real-things and a flight-simulator model via the Arduino. Figured out a few problems you might want to include in the next version to save others from head-scratching !!

  1/ ArduinoListner.ICO file .. 

   You've got a bug %line 127 for digitalPinToRead .. it should use data[2] instead of data[0].

  2/ In latest release - ArduinoDriver.cs - method InitializeWithoutAutoBootstrap()

  You have a port.Close() after the handshake... this shouldn't be there (creates load of problems)...  change the code to be 

  if (handshakeResponse == null)
 { 
    port.Close(); 
    Port.Dispose(); 
    throw exception.. 
}

 Cheers

Gary

christophediericx, 2017-01-22:
Hi Gary,

Thank you for your interest and for taking the time to report these issues.

The first one is very clearly a bug, I'll need to do some digging to have a look at the second issue.

In any case, I'm planning to get these issues resolved later this evening!

Thanks,

Christophe

christophediericx, 2017-01-23:
Hi Gary,

Thanks you for your bug report - both were spot on! The suggested changes have been pushed to the main branch. A new nuget package (2.1.1) will be available in a couple of minutes.

Cheers,

Christophe

closed 2017-01-23 by christophediericx