christophediericx/ArduinoDriver #11: External Analog Reference

DynastyORiGiN, 2017-03-03:
Please could you add "external analog reference" functionality to your program as without it the Arduino Uno R3 has very poor analog read accuracy as Vcc (default reference) varies.

I have looked into adding it in myself.  
The easiest way, I assume, would be a to add the command to ArduinoListener.ino file but I was unsuccessful in compiling the new required ArduinoListener.ino.UnoR3.hex due to Arduino IDE compiler errors.

The other option I believe is to add the command set to all required files (to the library) but again without the ability to generate the required hex file this would be all for nothing.

Thanks 
DynastyORiGiN

christophediericx, 2017-03-04:
Hi DynastyORiGiN,

Thank you for your interest in this library.

Adding external analog reference sounds like a very reasonable thing to do. At the moment I'm travelling, but I will likely have some time to add this feature in the coming week (March 6 - 10).

Best regards,

Christophe

christophediericx, 2017-03-11:
Hi DynastyOrigin,

I have implemented setting the Analog Reference now. Usage is very similar to sending other requests, e.g.

`
var response = driver.Send(new AnalogReferenceRequest(AnalogReferenceType.External));
`

Try it out, and feel free to reopen if you run into any issue!

Thanks,

Christophe

closed 2017-03-11 by christophediericx