christophediericx/ArduinoDriver #12: Add support for shiftout operation

Herbstein, 2017-03-21:


christophediericx, 2017-03-22:
Hi Herbstein,

Thank you for your interest in this library. I agree, this would be a useful feature indeed. I will try to implement it in the coming week.

Thanks,

Christophe

christophediericx, 2017-03-28:
Hi Herbstein,

Support for both `ShiftOut`  and `ShiftIn` operations have been added to the ArduinoDriver (version 2.4, you might need to update your referenced nuget package).

A sample call:

`var response = driver.send(new ShiftOutRequest(5, 2, BitOrder.MSBFirst, 0x13)`

Feel free to reopen the ticket if the functionality should not work correctly!

Christophe

closed 2017-03-28 by christophediericx

Herbstein, 2017-03-28:
Hi Christophe,

I didn't notice you had replied to the feature request I try and stay active in issues I open, so sorry about that!

I haven't been able to test the update yet (I've been away from my board) but it looks like a solid implementation. Thanks for the added support!

Herbstein

christophediericx, 2017-03-29:
Hi Herbstein, 

No worries, it can take me a couple of days to respond to issues, too.

Thanks,

Christophe