christophediericx/ArduinoDriver #13: Access to COM denied.

ProgenitorX, 2017-03-24:
Hello! Thanks for taking the time to write up this library. I'm having an issue I can't pin down. When I try to create an instance of ArduinoDriver I get an UnauthorizedAccessException stating it couldn't access the COM port. I ran your compiled sample code and it worked just fine. However, I tried compiling the source for the sample code and it receives the same error.

I'm using a UNOR3 via USB and VS 2015. Communications should be fine since your compiled sample worked flawlessly, but my test programs and me compiling your sample source does not. Let me know if you need additional information. Nothing else should have a hold on the COM port; I also tried manually entering the COM port using the overloaded constructor.

Another thing I noticed is that even though I have the ArduinoDriver Nuget Package installed and have access to the DLL, I need to specify the ArduinoDriver class using the namespace like so: "ArduinoDriver.ArduinoDriver". Adding "using ArduinoDriver;' doesn't remove this issue. I don't think it's a problem, just found it odd.

I'd like to test out your library and see if it fits my needs, but I can't get past the COM access. Please advise on what I should do to get past this. Thanks!

christophediericx, 2017-03-28:
Hello,

Thank you for your interest in this library. As I was investigating this issue this evening, I managed to reproduce what the likely culprit was in the "protocol upgrade" path. A tentative fix has now been put in place in the latest version of the code (a nuget package - ArduinoDriver 2.4 will be released, too).

Could you please try to update the code and see if this fixes the issue for you?

ProgenitorX, 2017-03-28:
Downloaded the version here and also the one on Nuget; both are working now. I did notice the LED flashes for the SMB Underground theme song (I don't have sound output) seemed to flash at a faster tempo than before. Just an observation. The serial connection went through and the board is writing and receiving the tones from the sample just fine now.

Thanks for looking into this!

christophediericx, 2017-03-28:
Thanks for trying it out, I'm happy it works correctly now.

The issue was the following: as new features get implemented in the driver at the C# side (e.g. new pairs of requests / responses), an updated version of the "listener" on the Arduino needs to be deployed. 

This version negotiation is a part of the protocol handshake (it was in this code path that the COM port close bug you encountered was located). So in other words, you will see the LEDs on your board flash (very intensely) for a couple of seconds (only once) *if* a protocol update is being automatically distributed: this is indicative of your board getting flashed with the most recent version of the listener.

closed 2017-03-28 by christophediericx