christophediericx/ArduinoDriver #15: Read and write multiple values with one request

Herbstein, 2017-03-28:
As it stands now reading and writing multiple values at once can be quite slow. Writing a digit to a 7-digit display takes around 85 ms. That is a long time taken to write output to seven pins. 

I propose that new options are added to the protocol. The names are unimportant so I'll refer to the additions as `ExtendedRead` and `ExtendedWrite`. These cover both analog and digital operations. 

The protocol extensions would be similar to the current implementations. The extensions would send the amount of operations (operation referring to reading or writing a pin) to the board, along with the needed pin information to complete the operations.

For reading this would simply be a number, eg. 3, and a list of the pins to read. An example could look like this:

`var response = driver.Send(new ExtendedDigitalRead(3, 4, 5);`

The only parameter to the `ExtendedDigitalRead` constructor would be a single `params` parameter containing all the pins.

Writing to multiple pins is slightly trickier. I suggest wrapping the pin and `DigitalValue` value in a struct.

`struct DigitalWriteInfo { byte Pin { get; set; } DigitalValue Value { get; set; } }`

The constructor is omitted from the above declaration. An example call with three values would then look like this:

`var response = driver.Send(new ExtendedDigitalWrite(new DigitalWriteInfo(2, DigitalValue.High), new DigitalWriteInfo(3, DigitalValue.Low), new DigitalWriteInfo(4, DigitalValue.Low)));`

The syntax is very verbose, and there might be a better way to handle it, but I believe the intention is clear.

Feel free to ask for clarification if you need it.

christophediericx, 2017-03-29:
That sounds like a reasonable request. Thanks for providing the write-up.

I'll try to make some time somewhere in the next week to implement this.

christophediericx, 2017-04-05:
Hi Herbstein,

As I had to travel for work, I will only have time to implement this next weekend. Sorry for the delay.

Thanks,

Christophe

Herbstein, 2017-04-05:
Hi Christophe.

That isn't a problem. I'm using the library for a project, but it's not that time sensitive. Thanks for the effort you're putting in!

Herbstein