christophediericx/ArduinoDriver #18: Question: The loop behavior from the Arduino IDE

vnvizitiu, 2017-04-15:
Hello,

I was wondering if there is any future plans for implementing the Arduino loop, maybe as a lamda expression or something?

Thank you,
Vlad V.

christophediericx, 2017-04-20:
Hi,

Thank you for your interest in this library.

I will try to have a look at your question the coming weekend.

Thanks,

Christophe

christophediericx, 2017-04-23:
I'm not sure if I understand the intent completely, could you please elaborate?

Can you not simply replicate the Arduino "loop" (conceptually) like this?

```
using (var driver = new ArduinoDriver(ArduinoModel.UnoR3, true))
{
	while (true)
	{
		// Implement methods like you would in the 'loop' method.
	}
}
```

vnvizitiu, 2017-04-23:
Hey,

To my shame i did not try that, (not knowing alot about the compiled arduino files) i though it was written in some other way so i can deploy it, if i run that in visual studio will the file stay deployed even if i close visual studio?

Thank you,
Vlad V.

christophediericx, 2017-04-23:
Hi Vlad,

I'll try to explain a bit how the ArduinoDriver works, that may be helpful in answering some of your questions.

The library does not compile down C# code to AVR assembly (that would be super neat but also very very hard from a technical perspective). It requires a persistent serial connection (over USB) between the running C# code and the Arduino device. 

What happens when you use the ArduinoDriver directly from C# code, is that an Arduino "Listener" (https://github.com/christophediericx/ArduinoDriver/blob/master/Source/ArduinoDriver/ArduinoListener/ArduinoListener.ino) gets autodeployed onto the Arduino device (by talking to the device's bootloader).

That listener at the Arduino will listen to incoming requests (as composed and sent by the C# library over the serial connection). It will execute the payload on the Arduino itself, and then formulate a response (which is being parsed out and transformed back into C# objects at the C# side).

It hope this clarifies things a bit.

Best regards,

Christophe

vnvizitiu, 2017-04-23:
Got it,

Thank you for the explanation, now i have a better grasp of what's happening,
Vlad V.

closed 2017-04-23 by christophediericx