christophediericx/ArduinoDriver #22: Servo motor functions

sampaisenpai, 2017-07-24:
Hello,

I would like to suggest adding a way to control servo motors with this library.

Usually, with an Arduino, one would simply use the "Servo.h" library to control a servo motor. However, when using this library, one must recreate the whole PWM functions since a servo motor is expecting a 40Hz frequency which isn't really obtainable by using analogWrite commands. 

I beleive it could help greatly to have such a function :) (if there already is one and I simply couldn't find it, I'm sorry and could you please point me in the right direction?)

dimakan, 2017-08-28:
Hi,
I also plan to use servo. 
I hope that ToneRequest or something similar will do the trick. 
Sounds like pwn command...
Please let me know, if there is better way for low level servo handling for now. 

Thanks,
Dima

var driver = new ArduinoDriver(ArduinoModel.UnoR3, true);
driver.Send(new ToneRequest(8, 200, 1000));

dimakan, 2017-08-29:
Found much better solution :)
Connecting full function secondary arduino allows any usage. 
First arduino connected by USB to pc provides control signal and secondary handle complex hardware with all libraries!
This is more flexible, as one cannot duplicate all the arduino libraries in c#. 

Hope it helps

Oluntridao, 2018-02-26:
i'm to trying to do the same think. I don't know how i can make servo control through arduinodriver.

mikalting, 2018-11-30:
> 
> 
> i'm trying to do the same think. I don't know how i can make servo control through arduinodriver.

have you find a way to control the servo motor?