christophediericx/ArduinoDriver #23: Arduino M0 Pro Support

DroneMan, 2017-07-26:
Any chance support for the Arduino M0 Pro could be added?

christophediericx, 2017-08-02:
Hi,

Thank you for your interest in this library, and sorry it took me some time to get back to you. 

It looks like the M0 **Pro** supports STK500v2 protocol style bootloader communication, which means support for it should be quite straightforward to add within the current code base. I don't have the relevant hardware to test however, so perhaps (once implemented) you could help test and see if the implementation actually works.

I'll try to find some time for this in the next weekend, and will update the ticket once the implementation is done.

Thanks,

DroneMan, 2017-08-03:
Excellent, I can certainly test the implementation once it's uploaded. Looking forward to it.
Cheers

christophediericx, 2017-08-10:
Sorry, I did not find the time last weekend. Next weekend should be good, though.

christophediericx, 2017-08-17:
Sorry for the delay. I had a quick look at this yesterday, but it seems trickier than initially anticipated. I should have time to implement this somewhere early September.