christophediericx/ArduinoDriver #30: LED HIGH example

mikalting, 2018-11-28:
hello I'm a noob who is trying to understand how the arduinodriver is working, but i can't find anything basic how the driver is working.
could some maby send me sort of a sample to turn on a led.

TehseenHasan, 2018-11-29:
If somebody make the example that will be very helpful...

mikalting, 2018-11-30:
> 
> 
> If somebody make the example that will be very helpful...

i use this:
using (var driver = new ArduinoDriver.ArduinoDriver(AttachedArduino, true))
            {             
                driver.Send(new DigitalWriteRequest(13, ArduinoDriver.DigitalValue.Low));
                driver.Send(new DigitalWriteRequest(13, ArduinoDriver.DigitalValue.high));
            }